Open-CMSIS-Pack/vscode-cmsis-debugger #34: Use gdbtarget `serverPortRegExp` setting for default connections

jreineckearm, 2025-02-24:
The `gdbtarget` debugger type of the CDT GDB Adapter extension has the `serverPortRegExp` server parameter. This allows to scan the debug console output for a log message with the default port of a GDB server. We should use this rather than having to set a default port explicitly.

We should aim for valid expressions to use in pyOCD and J-Link default configurations/auto-fill of configurations during debug launch.

omarArm, 2025-02-26:
Hi,

I managed to have a regex that would capture the port number.

now instead of the "port" option, we can use the following

`"serverPortRegExp": "[0-9]? I GDB server started on port ([0-9]+) \\(core 0\\).*\r?\n"`

jreineckearm, 2025-05-20:
Thanks again for the investigation. Let's keep this in mind in case we need it again in future. But at the moment the port management will happen through debug adapter templates managed by CMSIS Solution extension.

closed 2025-05-20 by jreineckearm