Open-CMSIS-Pack/vscode-cmsis-debugger #36: Allow in-session target resets with Arm Debugger extension

jreineckearm, 2025-02-24:
Currently, the CDT GDB adapter extension doesn't support a custom "restart" operation. This results in VS Code ending the entire debugger infrastructure just to restart it when pushing the "restart" button in VS Code. However, most embedded SW debugger offer the ability to use target hardware resets which are normally much faster.

We need to investigate how to be best contribute/extend the restart with custom behavior. Changes may be necessary in the CDT GDB adapter itself. It may be worth to investigate how a general concept could look like to override/extend debugger functionality with custom commands.

Note that for some operations, more than just issuing a single command via GDB may be necessary. A reset for example needs to send events to inform the IDE about the target systems state during/after the reset. Otherwise, a CPU may not have stopped and any subsequent attempt to call a command can end in failures.

Example: pyOCD offers in-session reset functionality via `monitor reset` commands.

Expectation is that changes are required in both
https://github.com/eclipse-cdt-cloud/cdt-gdb-vscode
https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter

To raise specific issues after initial brainstorming.

jreineckearm, 2025-04-25:
First prototyping proved feasibility of an approach to optionally contribute a button to the debug toolbar for in-session resets.
Created issue at CDT GDB adapter repository to drive this forward at the source: https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter/issues/359

In addition to that, we need updates in pyOCD to reflect the CPU state during reset and give the GUI a chance to catch changes. Proposal is to use GDB `*running` and `*stopped` asynchronous notifications. They are already supported by the CDT GDB adapter.

jreineckearm, 2025-04-25:
Should have mentioned before the last comment that we have agreed to NOT override the `restart` functionality but to optionally contribute a separate button. Rational is that there are valid reasons for keeping a "restart" in the way that VS Code offers it by default.

thorstendb-ARM, 2025-05-15:
See PRs:
- https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter/pull/365
- https://github.com/eclipse-cdt-cloud/cdt-gdb-vscode/pull/152

jreineckearm, 2025-05-16:
Almost done but shall not hold up 0.2.0 release with pyOCD supporting new configuration flow. Release preview updates as needed when ready. Moved to 1.0.0 timeframe.

jreineckearm, 2025-05-22:
Support in CDT GDB Adapter extension released with v2.0.4.
Requires fix for #166 in pyOCD for completion.

jreineckearm, 2025-06-04:
Feature as such finished. Closing to reflect no work remaining on extensions side. Remaining work to make this more seamless in pyOCD in #166 .