Open-CMSIS-Pack/vscode-cmsis-debugger #39: Verify E2E flow with selected boards

jreineckearm, 2025-02-25:
Once we have a first extension version in place, we need to test it with some real-world target systems.

The aim is to
* identify and capture user flow issues across all involved extensions and OSS components.
* raise those issues against the corresponding OSS projects.
* get a better idea of the current quality level of the overall solution.

The exact list of tested boards is TBD.

ReinhardKeil, 2025-06-26:
#Alif AppKit  Blinky_HE
## with JLink Server
- RTX awareness is missing
- GCC: seems to have fake breakpoints
## with CMSIS-DAP
- Reset not working (Segger seems to work)
- GCC: breakpoint issues (does not stop at main, breakpoints set before starting debug don't work)

jreineckearm, 2025-07-31:
Assigning to myself to close once all other issues have been validated. This should give enough confidence that the flows are working. Remaining caveats shall then be addresses separately.

jreineckearm, 2025-07-31:
Manual testing of the flow happened during release preparations. This included release candidates of all involved components:
* Demos of the above mentioned board from Alif were tested.
* Various boards from other silicon vendors have been used during exploratory testing and defect validation.

High confidence in the solution with the release candidates.
Any later found flaws to be raised and addressed separately for future releases.
Good to close the ticket.

closed 2025-07-31 by jreineckearm