Open-CMSIS-Pack/vscode-cmsis-debugger #65: [Feature Request]: Use Arm CMSIS Debugger with pyOCD for multi-core debug

jreineckearm, 2025-03-05:
### Description

We need to investigate how to utilize the solution with pyOCD to drive multi-core debug sessions.
That is connection strategies like
- connect to one CPU in the system, then subsequently connect others in the same.
- launch a connection to multiple CPUs in the system "simultaneously".

For the latter, VS Code has built-in mechanisms like "[compounds](https://code.visualstudio.com/docs/editor/debugging-configuration#_compound-launch-configurations)".

Also, the CDT GDB Adapter extension includes the [CDT Amalgamator](https://github.com/eclipse-cdt-cloud/cdt-amalgamator) which could be an alternative approach.

We should explore which approach is best balanced in terms of scalability, functionality, configurability, and user experience.

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

closed 2025-05-05 by jkrech

jkrech, 2025-05-05:
Out of the box dual core example (including launch scripts) - branch "cmsis-debugger"

    [NXP_FRDM-MCXN947 DualCore](https://github.com/Open-CMSIS-Pack/NXP_FRDM-MCXN947_BSP/tree/cmsis-debugger/boards/frdmmcxn947/cmsis/examples/DualCore)

    [FRDM-K32L3A6 DualCore](https://github.com/Open-CMSIS-Pack/csolution-examples/tree/cmsis-debugger/DualCore)

    [Alif Ensemble AppKit DualCore](https://github.com/VladimirUmek/alif_ensemble_appkit-bsp/tree/cmsis-debugger/examples/M55_HP_HE/DualCore)

    [STM32H745I-DISCO Blinky (Dual Core)](https://github.com/Open-CMSIS-Pack/STM32H745I-DISCO_BSP/tree/cmsis-debugger/Examples/Blinky)

KeilChris, 2025-05-09:
The example for FRDM-K32L3A6 is not working. It lacks some files:
- vcpkg_configuration.json file is missing for puling in gbd
- You should not put a manual cbuild-run.yml in the project, trust the one built with the latest tools.

But even if you fix these issues, it does not work for me:
```text
GNU gdb (Arm GNU Toolchain 14.2.Rel1 (Build arm-14.52)) 15.2.90.20241130-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=aarch64-apple-darwin20.6.0 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.linaro.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
warning: Loadable section "RW_RAM0" outside of ELF segments
  in /Users/chrsei01/02_Git/Open-CMSIS-Pack/csolution-examples/DualCore/out/HelloWorld_cm4/FRDM-K32L3A6/Debug/HelloWorld_cm4.axf
0000299 I Target type is k32l3a60vpj1a [board]
0000467 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0000472 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
0000473 I AP#1 IDR = 0x001c0040 (AP var4 rev0) [discovery]
0000474 I AHB-AP#2 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0000477 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4c4) [rom_table]
0000479 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0000479 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0000480 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0000481 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0000483 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0000485 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0000486 I [8]<e0044000:CTI CS-400 class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0000489 I AHB-AP#2 Class 0x1 ROM table #0 @ 0xf0002000 (designer=00e:NXP part=000) [rom_table]
0000491 I [0]<f0000000:MTB M0+ class=9 designer=43b:Arm part=932 devtype=31 archid=0a31 devid=0:0:0> [rom_table]
0000493 I [1]<f0001000:MTBDWT class=9 designer=00e:NXP part=000 devtype=04 archid=0000 devid=0:0:0> [rom_table]
0000495 I [2]<f0006000:CTI M0+ class=9 designer=43b:Arm part=9a6 devtype=14 archid=1a14 devid=1040800:0:0> [rom_table]
0000496 I [3]<e00ff000:ROM class=1 designer=43b:Arm part=4c0> [rom_table]
0000496 I   AHB-AP#2 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c0) [rom_table]
0000497 I   [0]<e000e000:SCS v6-M class=14 designer=43b:Arm part=008> [rom_table]
0000498 I   [1]<e0001000:DWT v6-M class=14 designer=43b:Arm part=00a> [rom_table]
0000499 I   [2]<e0002000:BPU v6-M class=14 designer=43b:Arm part=00b> [rom_table]
0000502 I CPU core #0: Cortex-M4 r0p1, v7.0-M architecture [cortex_m]
0000502 I   Extensions: [DSP, FPU, FPU_V4, MPU] [cortex_m]
0000502 I   FPU present: FPv4-SP-D16-M [cortex_m]
0000503 C Error: <APv1Address@0x110740490 #2 dp=0> [__main__]
gdbserver has exited with code 1
```

RobertRostohar, 2025-05-09:
There is a bug in the NXP K32L3A60_DFP 19.0.0 which incorrect AP description for Cortex-M0+. See more information in #94.

Even when using the latest cmsis-toolbox you still need to manually update the `cbuild-run.yml` as described in #94. This is the reason why in the example uses manual `cbuild-run.yml`.

KeilChris, 2025-05-09:
@RobertRostohar, you are correct! I overlooked that. So the only thing missing in the branch is the vcpkg config file.

RobertRostohar, 2025-05-09:
The vcpkg file was intentionally not added since there is already one in the root folder. Eventually this branch should go away once the DFP is updated with correct AP and since the `cbuild-run.yml` will be autogenerated together with the `launch.json` and `tasks.json`.