Open-CMSIS-Pack/vscode-cmsis-debugger #67: [Bug]: (Windows-only?) - Host encoding warning on first symbol request

jreineckearm, 2025-03-05:
### Summary

A warning is issued in the Debug Console which so far seems to be harmless

### Description

Whenever I establish a connection with pyOCD and `arm-none-eabi-gdb`, the first time request of a program symbol (e.g. for setting a breakpoint or showing a variable) provokes the following warning

`warning: could not convert 'symbol' from the host encoding (CP1252) to UTF-32.`

This hasn't been seen yet on for example MacOS.
The code issuing this warning is in GDB itself, i.e. the `arm-none-eabi-gdb` executable generates it.

While the warnings seems to be harmless so far, we should understand if it is provoked by pyocd and if we can do anything about it. It looks scary if you are not used to it.

It shows exactly once per debug session, that behavior can be confirmed by reading GDB source code.

### Steps to Reproduce

1. Launch a debugger connection with pyOCD and `arm-none-eabi-gdb`,
2. Provoke retrieval of a debug symbol (e.g. set a breakpoint on a function or explore the Variables view)
3. You should see this warning happening once per debug session. At least on Windows.

### Expected Behavior

Ideally we wouldn't see that warning.

RobertRostohar, 2025-09-02:
This seems to be an issue of the gdb client.

There are multiple reports about similar problems:
- [Bug 5993](https://bugs.linaro.org/show_bug.cgi?id=5993) - could not convert 'VAR_NAME' from the host encoding (CP1252) to UTF-32.
- [Bug 6070](https://bugs.linaro.org/show_bug.cgi?id=6070) - UTF-8 Charset not supported
- [Bug 32853](https://sourceware.org/bugzilla/show_bug.cgi?id=32853) - [gdb/symtab] warning: could not convert 'main' from the host encoding (ISO-8859-1) to UTF-32
- [Bug 29315](https://sourceware.org/bugzilla/show_bug.cgi?id=29315) - Completion warning: could not convert ... from the host encoding (ANSI_X3.4-1968) to UTF-32
- [Bug 29205](https://sourceware.org/bugzilla/show_bug.cgi?id=29205) - Running selftest dw2_expand_symtabs_matching. warning: could not convert 'yfunc�' from the host encoding (UTF-8) to UTF-32

jreineckearm, 2025-10-03:
Thanks for pointing this out. Indeed seems to be a problem with which locale is active on your Windows system.
Seeing similar issues with UTF8 characters in file names in makefile driven builds with gcc.
Nothing we should aim to fix. Yet, I'd like to spend a bit of time to find out how to overcome this (if possible) and to document/link a workaround.

closed 2025-11-12 by jreineckearm