Open-CMSIS-Pack/vscode-cmsis-debugger #94: [Bug]: Cannot connect to NXP FRDM-K32L3A6 with pyOCD

jreineckearm, 2025-03-11:
### Summary

As the title says

### Description

Any attempt to connect to the NXP FRDM-K32L3A6 board with the shipped pyOCD fails during auto-detection with:
```
0001183 I CPU core #0: Cortex-M4 r0p1, v7.0-M architecture [cortex_m]
0001183 I Extensions: [DSP, FPU, FPU_V4, MPU] [cortex_m]
0001183 I FPU present: FPv4-SP-D16-M [cortex_m]
0001184 C Error: <APv1Address@0x1e77c825b50 #2 dp=0> [__main__]

gdbserver has exited with code 1
```
That is with CMSIS-Pack debug support. There is no built-in support for this device in pyOCD.
I haven't managed yet to succeed with a connection with the J-LINK GDB server. However, didn't try hard yet.

### Steps to Reproduce

1. Try to connect to the board with pyOCD driven from the Arm CMSIS Debugger/CDT GDB Adapter solution.
2. Observe the error

### Expected Behavior

Connection, in particular auto-detection of this multi-core board, succeeds.

RobertRostohar, 2025-03-27:
NXP K32L3A6 device is a multi-core device and requires the correct debug port information to be specified in the `cbuild-run.yml`.

The enhancements to `cbuild-run.yml` for multi-core are still being reviewed and to be eventually generated by upcoming CMSIS-Toolbox releases. Current CMSIS-Toolbox 2.8.0 does not generate them.

pyOCD shipped in [v0.0.3 pre-release](https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/releases/tag/v0.0.3) already uses the proposed enhancements and successfully connects to K32L3A6 when the following is added to the `cbuild-run.yml` file:
```
  debug:
    debugports:
      - dp: 0
        accessports:
          - apid: 0
            index: 0
          - apid: 1
            index: 2
    processors:
      - pname: cm4
        apid: 0
      - pname: cm0plus
        apid: 1
```
Note that there is currently a mistake in the NXP K32L3A60_DFP 19.0.0 regarding AP description for the Cortex-M0+ processor. The AP index specified is 1 but the actual AP index is 2 (as shown above: `apid: 1`, `index: 2`,  `pname: cm0plus`.

Out of the box connection to NXP K32L3A6 will be working once:
- `cbuild-run.yml` enhancements for mulit-core devices are finalized.
- CMSIS-Toolbox supports those enhancements (currently they need to be added maually).
- NXP K32L3A60_DFP pack is updated with correct AP index for Cortex-M0+ (otherwise manual correction of the generated `cbuild-run.yml` is needed).
- pyOCD is updated if the enhancements specification changes (otherwise no change in pyOCD is needed).

Using pyOCD with multi-core devices and without providing debug port information is not desired. Even if pyOCD can auto-detect the processors, they cannot be linked to processor names (`pname`) which might be used for debug sequences, flash algorithms.

RobertRostohar, 2025-04-04:
pyOCD shipped in newer releases than v0.0.3 already uses the updated `cbuild-run.yml` spec which should be finalized now. 

`cbuild-run.yml` needs to contain the following description (if not yet generated by the CMSIS-Toolbox using the updated K32L36-_DFP):
```
  debug-topology:
    debugports:
      - dpid: 0
        accessports:
          - apid: 0
            index: 0
          - apid: 1
            index: 2
    processors:
      - pname: cm4
        apid: 0
      - pname: cm0plus
        apid: 1
```

closed 2025-04-04 by RobertRostohar