Open-CMSIS-Pack/vscode-cmsis-debugger #108: [Bug]: GDB Server terminates immediately with Infineon KIT_T2G-B-E_LITE and pyOCD

jenhua01, 2025-03-19:
### Summary

As the title says

### Description

- Target Device: KIT_T2G-B-E_LITE (V1)
- Device Family Pack T2G-B-E_DFP
- Board Support Pack: KIT_T2G-B-E_LITE_BSP

I tried to run a Blinky example from the Infineon board, TRAVEO T2G Body Entry Lite evaluation kit. After starting the debug session, the following error occurs and the GDB server is terminated at the very beginning:
```
0001679 C RW section does not follow RO section [__main__]
gdbserver has exited with code 1
```

### Steps to Reproduce

1. Open and build a Blinky example for the KIT_T2G-B-E_LITE board in VS Code.
2. Set up a debug session with pyOCD.
3. Connect the device and start the debugger.
4. Observe the error and exit the debug session.

### Expected Behavior

Run the debug session successfully.

jreineckearm, 2025-03-19:
Thanks for raising this, @jenhua01 , and for providing the relevant instructions for the reproducer.
Some investigation suggests that this error message comes from pyOCD. @RobertRostohar, could you please verify if this is the case. And why the section arrangement upsets it?

RobertRostohar, 2025-03-24:
I'm not able to build the Blinky project. Please attach the complete project (including the generated axf/elf).

jenhua01, 2025-03-24:
Here is the complete project: [T2G_E.zip](https://github.com/user-attachments/files/19429556/T2G_E.zip)

RobertRostohar, 2025-03-26:
Thanks for providing the project.

This is a problem in pyOCD which is too restrictive in checking the flash algorithm (does not take into account section alignment).

Will be fixed soon.

RobertRostohar, 2025-05-16:
Should be moved out of 0.2.0 (not addressed yet).

TeoMahnic, 2025-07-28:
Addressed in https://github.com/pyocd/pyOCD/pull/1795

jreineckearm, 2025-07-30:
@jthuangarm , can you please validate with the pre-release here: https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/releases/tag/v1.0.0-pre0
Note the we currently don't plan to push the pre-release for testing to marketplace.

jthuangarm, 2025-07-31:
The error is gone, and everything is working as expected! Thanks for all your help.

closed 2025-07-31 by jreineckearm