Open-CMSIS-Pack/vscode-cmsis-debugger #109: [Bug]: Error "Invalid `control` node in debug sequence `PrintDeviceInfo`" and "Segmentation fault"

jenhua01, 2025-03-19:
### Summary

As the title says

### Description

- Target Device: KIT_T2G-B-E_LITE (V1)
- Device Family Pack T2G-B-E_DFP v1.0.0
- Board Support Pack: KIT_T2G-B-E_LITE_BSP v1.0.0

I tried to run a Blinky example from the Infineon board, TRAVEO T2G Body "High" Lite evaluation kit. The debugging process can reach the main function despite an error appeared at the very beginning:
```
0001777 W invalid 'control' node in debug sequence 'PrintDeviceInfo' [cmsis_pack]
0001797 I Target type is cyt4bf8cds [board]
0001903 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
...
```
After pressing Continue (F5), the program executes a few steps, but fails to handle the timer interrupt properly in the main code. As a result, execution cannot continue, and the Signal Handler is called with the following error:
```
Program received signal SIGSEGV, Segmentation fault.
0x00000048 in ?? ()
```

### Steps to Reproduce

1. Open a Blinky example for the KIT_T2G-B-H_LITE board in VS Code.
2. Enable "CMSIS-Compiler" > "CORE" and "STDERR, STDIN and STDOUT" with "Custom" in Software Components.
3. Download the Infineon [Device Configurator](https://softwaretools.infineon.com/tools/com.ifx.tb.tool.deviceconfigurator).
4. Generate the missing source files using the configurator (see Additional Information).
5. Build the project.
6. Set up a debug session with pyOCD.
7. Connect the device and start the debugger.
8. Observe the error and the program cannot proceed.

### Expected Behavior

Run the entire code without errors.

### Additional Information

**Generate source code using Device Configurator**:
After downloading the Device Configurator, locate the executable file, e.g. `<YourUserName>\Infineon\Tools\InfineonDeviceConfigurator-5.20.0\resources\launch-tool.exe`. Open a terminal right here and type the following command:
```
launch-tool.exe -l "<PathToPacks>\Infineon\T2G-B-H_DFP\1.0.0\device-info\device-db\props.json","<PathToPacks>\Infineon\T2G-B-H_DFP\1.0.0\Libraries\core-lib\props.json","<PathToPacks>\Infineon\T2G-B-H_DFP\1.0.0\Libraries\mtb-pdl-cat1\props.json" -d "<PathToProject>\RTE\Device_Configurator\design.modus"
```
Remember to change the absolute paths (`<PathToPacks>` and `<PathToProject>`) to locate all the files. A GUI will show up. Simply press "CRTL+S" to save the `design.modus` file. Meanwhile, all the missing source files will be generated.

If no device is found when the target board is connected, enable the Manage All Devices setting:
- Open the settings of the ARM Device Manager VS Code extension.
- Find the Device-manager: Manage All Devices setting and select its checkbox.

jenhua01, 2025-03-19:
Here I have attached the source files generated by the Device Configurator, so that you can skip steps 3 and 4 for reproducing the issue.
[Device_Configurator.zip](https://github.com/user-attachments/files/19348422/Device_Configurator.zip)

RobertRostohar, 2025-03-24:
I'm not able to build the project (using the attached Device_Configurator.zip). I get build errors. Please attach the complete project (including the generated axf/elf).

jenhua01, 2025-03-24:
Here is the complete project: [T2G_H.zip](https://github.com/user-attachments/files/19429570/T2G_H.zip)

RobertRostohar, 2025-03-26:
Thanks for the project and description.

The warning `invalid 'control' node in debug sequence` is issued by pyOCD native debug sequence processing using debug sequences described in the device pack (.pdsc). It seems that pyOCD does not handle correctly a `control` node without `if` or `while`. Such a block is not executed and a warning is issued.

The latest pyOCD provided with the [v0.0.3](https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/releases/tag/v0.0.3) pre-release already uses debug sequences provided from the cbuild-run.yml and should not have this problem. Please try it out.

There might be other issues when debugging this project (we don't have a board to try it out). But at least the mentioned debug sequence should execute correctly.

jreineckearm, 2025-03-26:
The v0.0.3 pre-release has been published earlier today to the VS Code marketplace and OpenVSX.
You should be able now to update through the VS Code extension manager.

jenhua01, 2025-04-01:
Thank you very much for the explanation and the reminder, @RobertRostohar and @jreineckearm .

After updating to v0.0.03, the warning `invalid 'control' node in debug sequence` is gone! However, a new error occurred after starting debug session:
```
0000952 I [1]<e0041000:ETM M7 class=9 designer=43b:Arm part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0000954 I [2]<e0042000:CTI CS-400 class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0000954 C Error: <APv1Address@0x1a968104350 #1 dp=0> [__main__]
gdbserver has exited with code 1
```
This error interrupts the GDB server, preventing from reaching the main() function. Any ideas or suggestions on how I might troubleshoot this issue?

jenhua01, 2025-04-01:
Great, thanks! I have modified cbuild-run.yml with your suggested solution because this board contains 3 cores in total, and it works for me now. Here I attached once again the remaining warning messages written in Description with more information:

```
...
0000969 I GDB server started on port 3335 (core 2) [gdbserver]
0007614 I Client connected to port 3333! [gdbserver]
warning: Loadable section "RW_RAM_DATA" outside of ELF segments in <ProjectFolder>\blinky\out\blinky\CM0p\blinky.axf
0x00007196 in ?? ()
0007626 I Attempting to load RTOS plugins [gdbserver]
connected to remote target localhost:3333
load ./out/Blinky/CM0p/Blinky.hex
...
...
Program received signal SIGSEGV, Segmentation fault.
0x00000048 in ?? ()
```
I hope this will help you to analyze this problem.

RobertRostohar, 2025-04-10:
GDB only sets the PC and has no knowledge about setting MSP.

The bootloader needs to properly set the MSP and PC for the application which it does. The problem is that the debug sequence ResetSystem stops the bootloader before it sets the MSP and PC (jumps to the application). Afterwards gdb modifies the PC but the MSP is wrong - see my previous explanation for more details.

Can you share your modification of the debug sequences?

jenhua01, 2025-04-11:
I apologize for the mistake I made. The changes in the .pdsc file were not actually applied. Since I had renamed the cbuild-run.yml file to add `debug-topology` in it for testing and didn’t realize that, the changes in the .pdsc file wouldn't be reflected in that renamed cbuild-run.yml.

The good news is that Scenario 1 now works as expected! If there's nothing else I should know about my changes, I will inform Infineon to update the .pdsc file accordingly. And of course you can go ahead and close this issue.

I really appreciate your support and your time.

closed 2025-04-14 by RobertRostohar

RobertRostohar, 2025-04-14:
@jenhua01, I had another look at the `ResetSystem` debug sequence in the file that you have attached. The handling when no valid application is found should be reworked. I suggest to enable VectorCatch and then Reset via AIRCR. Otherwise, the core will not stop. Please try out this scenario.

jreineckearm, 2025-04-14:
Please note that debug sequences normally try to separate configuration of the reset catch from the actual reset sequence. That way, you do not have to re-implement for each reset type.
See for example https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/dbg_debug_sqns.html#dbg_sqns_boot
The sequences to deal with reset catches are
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#resetCatchSet
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#resetCatchClear

A typical sequence usage during reset can be found here: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#usage_of_sequences
Look out for "Reset Device".

jenhua01, 2025-04-15:
Thank you for pointing this out, which is really helpful! I'll look into refactoring my current ResetSystem debug sequence to follow this approach, using ResetCatchSet and ResetCatchClear as appropriate.

RobertRostohar, 2025-10-02:
@jthuangarm, do you have the latest version of the debug sequences in regard to Reset? Has this been shared with Infineon?