Open-CMSIS-Pack/vscode-cmsis-debugger #130: [Bug]: Setting more than hardware available breakpoints fails

TeoMahnic, 2025-03-26:
### Summary

pyOCD and gsdbserver are unresponsive fi you set more than hardware available breakpoints. 

### Description

If you try to set more breakpoints during a debug session than there are available hardware breakpoints, communication between gdbserver and pyOCD is lost.

### Steps to Reproduce

1. Start debug session.
2. Set all available hardware breakpoints and add one more.
3. Observe: pyOCD and gdbserver become unresponsive.

jreineckearm, 2025-05-16:
Not a blocker for 0.2.0 release. Move out to 1.0.0 timeframe.

DavidLesnjak, 2025-06-23:
I’ve traced the root cause of this behavior to how the CDT-GDB-Adapter’s MI parser handles the two-stage response from GDB when a hardware breakpoint fails. Below is a step-by-step reproduction and analysis:

**1. Test setup**
Target only supports 8 hardware breakpoints.
In your VS Code debug session, set 9 breakpoints in your source.

**2. pyOCD’s raw behavior**
Wireshark shows eight successful `$Z1` packets for breakpoints 1–8 _(wireshark_log.txt, lines 67–82)_.
The ninth `$Z1` is refused by the target _(line 84)_, as expected.

**3. GDB’s “all-stop” behavior**
In MI all-stop mode, GDB accepts all nine breakpoints internally without consulting pyOCD _(log.txt L181–L200)_ - it runs internal breakpoint list, that is propagated to pyOCD on command `-exec-continue`.
On `-exec-continue` _(log.txt L210)_, GDB first replies `^running` to the adapter, then sends eight `$Z1` requests to pyOCD (successful) and a ninth `$Z1` (which fails). Because of that failure, GDB emits a final `^error,msg="Command aborted."` and halts the `continue` _(log.txt L227)_.

**4. Where the adapter goes wrong**
In MIParser.ts’s handleLine() (inside the CDT-GDB-Adapter), the code sees the first `^running` response tied to the `-exec-continue` token and immediately does:
`delete this.commandQueue[token];`
That deletion signals “we’re done with this command”, so when the later `^error,msg="Command aborted."` arrives, there’s no matching entry and adapter never dispatches the error back to the VS Code front end.

As a result, VS Code never learns that the `continue` failed.

Log Files:
[log.txt](https://github.com/user-attachments/files/20861581/log.txt)
[wireshark_log.txt](https://github.com/user-attachments/files/20861580/wireshark_log.txt)

RobertRostohar, 2025-07-15:
I believe that GDB behavior is not really a problem. Looking at the log (when using all-stop mode):
- all breakpoints that are set are accepted by GDB
- when trying to run GDB aborts the command and reports an error: Could not insert hardware breakpoints: You may have requested too many hardware breakpoints/watchpoints.

The CDT GDB adapter however does not handle properly the aborted run command.

jreineckearm, 2025-07-15:
Funny behavior was observed without the adapter. Let me clarify with @omarArm while he's still available today.

jreineckearm, 2025-07-30:
@omarArm , @thorstendb-ARM , can you please give this a try with the CDT GDB adapter preview VSIX I shared internally. Please close if you are satisfied with it.

jreineckearm, 2025-07-31:
Going to prepare a first CDT GDB adapter release for our CMSIS Debugger 1.0.0 release without this change. If only to have the change in an isolated adapter release later today or tomorrow.

The solution works nicely in all-stop mode, but hasn't been tested much in non-stop mode yet. Hence, high risk of breaking other adapter use cases. Alternatively considering a guard to apply only in all-stop mode which also requires another round of testing which would delay the release.

jreineckearm, 2025-09-05:
Finally completed tests, likely to be released in CMSIS Debugger 1.1.0 timeframe.
Also becomes relevant for data breakpoint support where even less HW resources are available. They can provoke the same problems.

jreineckearm, 2025-09-08:
Merged, planned to be released with next CDT GDB adapter release (currently at v1.2.0).

closed 2025-09-08 by jreineckearm