Open-CMSIS-Pack/vscode-cmsis-debugger #156: [Bug]: Multiple projects in one workspace problem

KeilChris, 2025-04-11:
### Summary

I can only use the last opened folder when adding multiple folders to the same workspace in VS Code.

### Description

I added multiple csolution-based project folders to a workspace in VS Code. While the folders are shown in the Explorer view, I can only run and debug the project in the folder that I added last. There's no way to tell VS Code to use the launch.json file from another project. This makes the "multi-project workspace" unusable.

### Steps to Reproduce

1. Open a csolution-based folder in VS Code.
2. In the Explorer view, right-click the view and select "Add Folder to Workspace..." and add another csolution project folder.
3. Try to run and debug the first project - it is not possible. All information will be taken from the last opened project's `.vscode` folder.

### Expected Behavior

Would like to have an option to select something like an "Active Project" that would use the information from the specified project's `.vscode` folder (and `vcpkg_configuration.json` file as well).

KeilChris, 2025-04-11:
Using the "Debug" button from the CMSIS view ends up with this error message:

[image]

Note: I was not able to select the correct `launch.json` anyway. It presented me with only one (which was the wrong one).

KeilChris, 2025-04-11:
Starting debugging from the "Run and Debug view", I was able to select the right `launch.json` file, but the debugger chose to flash the wrong image (from the other project). Did not work as well.

jreineckearm, 2025-04-11:
Needs review together with other involved extensions to understand capabilities of all involved parties. And how a seamless E2E flow looks like (command contexts, etc). Note that this is not limited to a review of the CMSIS Debugger and CMSIS Solution extensions. But also includes referenced Eclipse CDT.cloud extensions.

ReinhardKeil, 2025-04-23:
Will be addressed with https://github.com/ARM-software/vscode-cmsis-csolution/issues/227

jreineckearm, 2025-05-07:
Can't remember that I moved this to `0.2.0`. Assume this is because mentioned CMSIS Solution extension ticket is in progress now?

jreineckearm, 2025-05-09:
Assigning to myself to keep track of what's going on in `vscode-cmsis-csolution` ticket. For the time being not expecting work on the debugger.

jreineckearm, 2025-05-13:
Move to 1.0.0, this should not block 0.2.0 release as it doesn't require work in the debugger extension.

ReinhardKeil, 2025-05-21:
This will be solved in CMSIS solution extension