Open-CMSIS-Pack/vscode-cmsis-debugger #164: [Bug]: Detach and Attach to the same core does not work

ReinhardKeil, 2025-04-15:
### Summary

Detach and Attach to the same core does not work

### Description

Target: NXP_FRDM-MCXN947 [DualCore](https://github.com/Open-CMSIS-Pack/NXP_FRDM-MCXN947_BSP/tree/cmsis-debugger/boards/frdmmcxn947/cmsis/examples/DualCore)

[image]

I can attach to Core1 without problem.  But when I detach the debugger and attach it again it fails.

ReinhardKeil, 2025-05-07:
@RobertRostohar FYI

RobertRostohar, 2025-05-12:
@omarArm, thanks for the research. It seems that the `--persists` option provides what is needed. We will add this command line option to the launch/tasks template scripts.

closed 2025-05-12 by RobertRostohar