Open-CMSIS-Pack/vscode-cmsis-debugger #166: [Bug]: monitor reset command not working

ReinhardKeil, 2025-04-16:
### Summary

monitor reset command not working

### Description

Target [NXP_FRDM-MCXN947 DualCore](https://github.com/Open-CMSIS-Pack/NXP_FRDM-MCXN947_BSP/tree/cmsis-debugger/boards/frdmmcxn947/cmsis/examples/DualCore)

jreineckearm, 2025-04-16:
For pyocd, this may work better with
```
>monitor reset halt
```

The pyocd specific `halt` part instructs the debugger to enable a reset catch and to halt the CPU after reset.
But you may run into other issues that we need to look at. In general, `monitor` commands have a certain disconnect from the GUI.

Actions:
- Investigate if there are GDB events/notifications pyOCD can send that the CDT GDB Adapter can translate into Microsoft DAP events to refresh the GUI.

thorstendb-ARM, 2025-04-17:
Implemented a configurable reset button in cdt-gdb-vscode and cdt-gdb-adapter:
```
    "configurations": [
        {
            "name": "pyOCD cdt-gdb-adapter",
            "resetDeviceCommands": [
                "monitor reset halt hardware",
            ],

```
See:
https://github.com/thorstendb-ARM/cdt-gdb-vscode-tdb/pull/1
https://github.com/thorstendb-ARM/cdt-gdb-adapter-tdb/pull/1

This works but needs the above described single step to update GUI.

GDB supports sending async commands that are presumably triggered by the debugger (e.g. pyOCD). This would be needed to tell the GUI the debugger has executed the reset sequence and stopped (if configured so) the device, e.g. via GDB async command to GUI "stopped". This is sent by GDB on breakpoint hit, step, ...

Question:
Which command has to be sent from pyOCD to GDB, so that GDB sends (e.g.) "stopped" to the GUI (gdb-adapter).

jreineckearm, 2025-04-23:
> Which command has to be sent from pyOCD to GDB, so that GDB sends (e.g.) "stopped" to the GUI (gdb-adapter).

Looking at the CDT GDB Adapter implementation and possible [async records](https://sourceware.org/gdb/current/onlinedocs/gdb.html/GDB_002fMI-Async-Records.html#GDB_002fMI-Async-Records) in GDB/MI, pyOCD should send appropriate `*running` and `*stopped` async execution records for any monitor command that alters the CPUs execution state.

jreineckearm, 2025-05-16:
Move out to 1.0.0 release timeframe. CDT GDB Adapter changes still in the works and 0.2.0 release needs to go out to demonstrate new configuration flow with CMSIS Solution extension.

thorstendb-ARM, 2025-05-21:
IMHO, pyOCD should send an async update request for every action it executes or becomes aware of, regardless of it's source:
- GUI
- GDB
- `monitor` commands
- Target MCU Reset / Power Cycle
 
1. Command `>monitor reset` sometimes brings the UL+ in an undefined state where it cannot recover from (power cycle reset for UL+ required).
2. When in gdb/GUI STOP mode, `>monitor reset` starts the target without letting the GUI know.
3. Command `>monitor reset` does not hit breakpoints prev. set in the GUI.
4. Command `>monitor halt` stops the target after `>monitor reset` without letting the GUI know
5. Doing a `>step` after 4. GUI gets confused (shows "running") and does not react anymore (except "leave" session
6. After 5., a `>monitor reset` prints "Evaluation of expression without frameId is not supported." to GDB console
7.

thorstendb-ARM, 2025-05-21:
Other comments:
- should all error messages be duplicated to popups? 
[image]
[image]

-

ReinhardKeil, 2025-05-21:
> Other comments:
> 
> * should all error messages be duplicated to popups?
>   [image]
>   [image]

When using the command line, these additional pop up windows are not really helpful. Just distract the user.  Output just one message line (not two as shown in the images) is sufficient.

jreineckearm, 2025-05-21:
Please open a separate issue for this. Needs fixing outside pyocd.

thorstendb-ARM, 2025-05-21:
> Please open a separate issue for this. Needs fixing outside pyocd.

Done, see https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/issues/271

RobertRostohar, 2025-06-23:
> Looking at the CDT GDB Adapter implementation and possible [async records](https://sourceware.org/gdb/current/onlinedocs/gdb.html/GDB_002fMI-Async-Records.html#GDB_002fMI-Async-Records) in GDB/MI, [...]

I don't believe this is possible.

gdb communicates with the target debugger (pyOCD, J-Link) which is running the GDB server via the [Remote Serial Protocol](https://sourceware.org/gdb/current/onlinedocs/gdb.html/Remote-Protocol.html#Remote-Protocol).

[Notification Packets](https://sourceware.org/gdb/current/onlinedocs/gdb.html/Notification-Packets.html#Notification-Packets) that the target debugger can send are very limited. Basically only "Stop" is supported.

gdb initially checks if the target is halted with '?' packet. Afterwards gdb controls the target execution. 

gdb resumes target execution via 'vCont' packet (continue, step). In All-Stop-Mode, the communication is blocked until the target is stopped and the 'Stop Reply' packet is sent by the target debugger. In Non-Stop mode 'OK' packet is sent to acknowledge that the target is running and once the target is stopped the asynchronous 'Stop Reply' packet is sent.

gdb assumes:  
 - target initial state is determined via '?' packet
 - target is running after gdb issued resume, until 'Stop' packet is received
 - target can only be resumed from gdb (no external events as for example reset)

This means that if the target was stopped (gdb was notified via 'Stop'), gdb assumes that the target is halted. There is no event that would tell gdb that the target started to run (for example when the target is reset externally).

jreineckearm, 2025-06-23:
Thanks for the investigation. Need to further investigate if we can model such behavior differently, e.g. by somehow polling the current target state. Worst case through another monitor command or custom event that bypasses GDB (if possible at all).

jreineckearm, 2025-07-28:
Tried a couple of things including GDB commands supposed to refresh GDB's state (`info registers`, `info threads`, `maintenance flush` variants) and sending Stopped Events from CDT GDB adapter. But I am quickly running into situations where GDB gets confused by exiting threads on resets for yet not understood reasons.
That is for pyOCD. I haven't investigated Segger J-LINK yet.

The only stable approach I found so far is to append the reset commands to run to `main`.
```
"monitor reset halt",
"tbreak main",
"continue"
```

But this would only be a temporary solution for the upcoming release until we understand what's happening to GDB in the mentioned scenarios.

To stop execution before `main`, users have to set breakpoints for now.

jreineckearm, 2025-07-29:
Workaround in place for 1.0.0. Needs another look at after release. Keeping ticket in review and assigned to me until new ticket is created.

thorstendb-ARM, 2025-07-31:
Tested with NUCLEO-L476RG Blinky, works as expected: After RESET, program returns to main.

_Note: Works while program stops outside any RTX thread, the issue where gdb exits with assertion still exists._