Open-CMSIS-Pack/vscode-cmsis-debugger #167: [Feature Request]: Extending Disassembly View with context menu

ReinhardKeil, 2025-04-16:
### Description

Is it possible to extend the disassembly view with context menu.

uVision offered features like run to cursor line
There should be also a way to change the address for the disassembly window.

Such features could be enabled using a context menu in the disassembly view.

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

jreineckearm, 2025-04-16:
This https://github.com/microsoft/vscode/pull/212500 was merged only last week which adds a context menu to the disassembly view.

@thegecko, would this also allow to contribute commands to the context menu? Or would this need more work in VS Code itself?

Other actions
* Investigate which commands are available today in VS Code to control the position of the view.
* Double-check if breakpoints in the disassembly view currently work with our solution. IIRC, they need `supportsInstructionBreakpoints` support on MSFT DAP level, which I can't see set in the CDT GDB Adapters initialize response.

jreineckearm, 2025-04-16:
The merged VS Code PR is scheduled for the [April 2025 milestone](https://github.com/microsoft/vscode/milestone/306). I expect the enhancement to be released in early May.

thegecko, 2025-04-16:
My merged PR only adds the context menu (with a couple of commands)

This PR needs rebasing and merging before it can be extended: https://github.com/microsoft/vscode/pull/212501

ReinhardKeil, 2025-06-05:
I looks like the disassembler now supports a context menu.

This are the features available in uVision.  I suggest to implement the features in the red box first before investigating further.

[image]