Open-CMSIS-Pack/vscode-cmsis-debugger #168: [Bug]: Breakpoints set in disassembly view are ignored

ReinhardKeil, 2025-04-16:
### Summary

Breakpoints set in disassembly view are ignored

### Description

A breakpoint that has been set in disassembly view is shown in the Breakpoints view, but ignored.

[image]

### Steps to Reproduce

1. Set a breakpoint to a function in source code.
2. Open disassembly view and set a breakpoint on a interesting point (in my example a function call)
3. Remove the breakpoint set under (1)
4. Run application

jreineckearm, 2025-04-16:
IIRC, CDT GDB Adapter needs `supportsInstructionBreakpoints` support on MSFT DAP level, which I can't see set in the CDT GDB Adapters initialize response. Unfortunately, VS Code doesn't suppress setting breakpoints if that feature is not announced by the debug adapter.

jreineckearm, 2025-05-08:
@omarArm , now that you are familiar already with the breakpoint management in the CDT GDB adapter, maybe you could start thinking about how to add this support?

jreineckearm, 2025-05-20:
Created https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter/issues/373 to support this in the CDT GDB adapter.

ReinhardKeil, 2025-06-05:
Source and Disassembly view are not synchronized in the breakpoint management. In partice:
 - A breakpoint set in Source (Editor) view is not shown in Disassembly view
 - A breakpoint set in Disassembly view is not shown in Source (Editor) view
 - A breakpoint set in Disassembly view on a line number should be reflected in BREAKPOINTS list as it would have been set in Source View on a line number.
 
[image]

omarArm, 2025-06-24:
I managed to add support for instruction breakpoints insertion/deletion via the GUI. However, we are still missing a few things when it comes to using the CLI for setting or removing instruction breakpoint.

Please check this issue for reference https://github.com/microsoft/vscode/issues/252294

jreineckearm, 2025-07-02:
First step to enable breakpoints in the disassembly view is taken with CDT GDB Adapter release v2.0.6. Breakpoints toggled in the VS Code disassembly view are now correctly installed in target system.

However, there are still some limitations which cannot be overcome easily at the moment:
* Disconnect between VS Code breakpoint management and breakpoint management from GDB CLI. See above comment.
* No synchronization between source editor and disassembly view.

jreineckearm, 2025-07-14:
Can't close because more is asked in this issue than what has been done so far. But we are blocked as per above comment.
Reschedule as it's not realistic to further proceed before 1.0.0 release.

closed 2025-08-01 by jreineckearm