Open-CMSIS-Pack/vscode-cmsis-debugger #181: [Feature Request]: RTOS Viewer

jreineckearm, 2025-04-24:
### Description

Review existing RTOS viewer extensions for VS Code to include into Arm CMSIS Debugger solution.
Identify required changes in other involved extensions to make a suitable RTOS viewer work.

Main focus should be on OSS extensions.

### Is this request a Security Requirement?

- [ ] Yes

### Priority

Medium

jreineckearm, 2025-05-05:
One option of an RTOS viewer is https://marketplace.visualstudio.com/items?itemName=mcu-debug.rtos-views

If we should decide for this, then there are some issues to resolve first:
* Support for a dependency (`mcu-debug.debug-tracker-vscode`) of that extension is broken in Peripheral Inspector. Issue captured as https://github.com/eclipse-cdt-cloud/vscode-peripheral-inspector/issues/63
* Impact of `mcu-debug.debug-tracker-vscode` on other debug tracker implementations needs to be reviewed.
* CDT GDB Adapter expression evaluation currently fails due to incorrect resolution of members for object pointers. Issue not raised yet.
* Extension does not support RTX5.
* Extension could be a good start. But has room for improvement when it comes to UX.

jreineckearm, 2025-07-09:
Peripheral Inspector issue was a problem with how `mcu-debug.debug-tracker-vscode` is integrated there.
Reviewed the `debug-tracker-vscode`: Should keep an eye on how this behaves with other extensions and their trackers. But so far couldn't see major side effects.

thorstendb-ARM, 2025-07-15:
Current state: https://github.com/thorstendb-ARM/rtos-views-tdb/pull/1

jreineckearm, 2025-10-22:
> If this feature request scope is fulfilled with the work you did, could you please create another feature request which gains the same detailed functionality for RTX as it is in uVision?

@saperlot , thanks for your comment. In the background, we are also working on #187 which would bring the RTX5 [Component Viewer](https://developer.arm.com/documentation/101407/0543/Debugging/Debug-Windows-and-Dialogs/Component-Viewer) support based on SCVD descriptions.

Is this the one you are referring to? If you you have something else in mind, please feel free to raise a separate feature request.

saperlot, 2025-10-22:
> [@saperlot](https://github.com/saperlot) , thanks for your comment. In the background, we are also working on [#187](https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/issues/187) which would [...]

That's the one, yes... Great work!