Open-CMSIS-Pack/vscode-cmsis-debugger #218: [Feature Request]: How to support multiple program files for run and debug

jreineckearm, 2025-05-08:
### Description

The `gdbtarget` type has a `program` setting which seems to be expected to receive multiple files in a comma-separated string as per recent [proposals](https://github.com/ReinhardKeil/cmsis-toolbox/blob/main/docs/CMSIS-DebuggerIntegration.md#launchjson-and-tasksjson-update-process) (see `%symbol-file-list`). This is not the case.

We need an investigation to find out how to best pass in such lists. Could be
* updating the CDT GDB adapter to accept such string.
* expand this into `initCommands` list. There are commands for separately loading files into target memory and debug symbols into the debugger.

Depending on the outcome, changes need to go into the template files or into the CDT GDB adapter. 

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

jreineckearm, 2025-05-09:
Assigning to myself and @omarArm as we have both some pre-knowledge in this area.
@omarArm feel free to spend some thought on this. But we should only take action after discussing thoughts next week.

jreineckearm, 2025-05-16:
Not realistic anymore to get investigation results into 0.2.0 release. Address for first production quality release 1.0.0.

ReinhardKeil, 2025-05-21:
The templates solve that.

closed 2025-05-21 by ReinhardKeil