Open-CMSIS-Pack/vscode-cmsis-debugger #241: [Bug]: pyOCD does not use default values for CMSIS_PACK_ROOT

ReinhardKeil, 2025-05-15:
### Summary

pyOCD does not use default values for CMSIS_PACK_ROOT

### Description

Currently, pyOCD does not use the default values of CMSIS_PACK_ROOT
https://open-cmsis-pack.github.io/cmsis-toolbox/installation/#environment-variables

This creates a problem when systems have no setting for this variable.

Workaround is to add this variable in the envrionment.

TeoMahnic, 2025-05-16:
Addressed in: [v0.0.0-standalone-cbuild8](https://github.com/MatthiasHertel80/pyOCD/releases/tag/v0.0.0-standalone-cbuild8)

jreineckearm, 2025-05-16:
Tested on Windows, @KeilChris tested on macOS. Good to close on behalf of @ReinhardKeil .

closed 2025-05-16 by jreineckearm