Open-CMSIS-Pack/vscode-cmsis-debugger #242: [Bug]: GDB crashes on step over in RTOS Blinky

JonatanAntoni, 2025-05-15:
### Description

On MacOS pyOCD crashes on stepping thru a RTOS Blinky example, such as the Nucleo-F756ZB:

```
Temporary breakpoint 1, main () at ./Nucleo-F756ZB-Blinky/STM32CubeMX/NUCLEO-F756ZG/STM32CubeMX/Src/main.c:105
105	  HAL_Init();
[New Thread 536939760]
[Thread 1 exited]
[New Thread 536972940]
[New Thread 536973020]
[Switching to Thread 536939760]

Thread 2 hit Breakpoint 3, app_main_thread (argument=<optimized out>) at ./Nucleo-F756ZB-Blinky/Blinky.c:86
86	  printf("Blinky example\n");
12
warning: (Internal error: pc 0x8006b00 in read in CU, but not in symtab.)
7
warning: (Internal error: pc 0x8005a7e in read in CU, but not in symtab.)
warning: (Internal error: pc 0x80059b0 in read in CU, but not in symtab.)
8
warning: (Internal error: pc 0x8005a7e in read in CU, but not in symtab.)
2
warning: (Internal error: pc 0x8006b01 in read in CU, but not in symtab.)
2
warning: (Internal error: pc 0x8006b02 in read in CU, but not in symtab.)
warning: (Internal error: pc 0x8006b00 in read in CU, but not in symtab.)
5
warning: (Internal error: pc 0x8006b01 in read in CU, but not in symtab.)
warning: (Internal error: pc 0x8006b00 in read in CU, but not in symtab.)
7
warning: (Internal error: pc 0x8006b01 in read in CU, but not in symtab.)
[New Thread 2]
/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/src/binutils-gdb--gdb/gdb/infrun.c:6706: internal-error: finish_step_over: Assertion `ecs->event_thread->control.trap_expected' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
----- Backtrace -----
Backtrace unavailable
---------------------
/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/src/binutils-gdb--gdb/gdb/infrun.c:6706: internal-error: finish_step_over: Assertion `ecs->event_thread->control.trap_expected' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]

This is a bug, please report it.  For instructions, see:
<https://bugs.linaro.org/>.

/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/src/binutils-gdb--gdb/gdb/infrun.c:6706: internal-error: finish_step_over: Assertion `ecs->event_thread->control.trap_expected' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
0008078 I Client disconnected from port 3333! [gdbserver]
```

### Steps to Reproduce

1. Create new solution from blinky example
2. Launch debug session
3. Try to single step/step over or run multiple times using some brakepoints.
4. GDB crashes with above log output

RobertRostohar, 2025-05-16:
It seems that you are using an image with DWARF 4 symbols. gdb has certain issues with that and DWARF 5 should be used. See [here](https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger?tab=readme-ov-file#arm-none-eabi-gdb-requires-dwarf5-debug-information).

Can you please add the following to your solution and try to see if the problem disappears:
```
  misc:
    - for-compiler: AC6
      C-CPP:
        - -gdwarf-5
      ASM:
        - -gdwarf-5
      Link:
        - --entry=Reset_Handler
```

jreineckearm, 2025-05-16:
In addition to that, I believe you are seeing the same as #159 where GDB exits, presumably because of a HW defect in early Cortex-M7 revisions that shows during stepping. See linked issue for details. Can you please check the CPUID register value?
https://developer.arm.com/documentation/ddi0489/f/system-control/register-summary
https://developer.arm.com/documentation/ddi0489/f/system-control/register-descriptions/cpuid-base-register

KeilChris, 2025-05-16:
I have tried with DWARF5 but still no luck (due to the HW IP error). I can confirm that the core revision on the STM32F756ZG is indeed affected by the error (r0p1).

RobertRostohar, 2025-05-26:
This is related to the Cortex-M7 r0p1 issue as described in #159.

There is workaround for STM devices when the peripheral interrupts are triggering this. Peripherals can be automatically stopped when the core is halted. This can be achieved by configuring the Debug MCU APB freeze registers.

In the above example, TIM14 is being used as a system tick and running periodically. Setting bit 8 (DBG_TIM14_STOP) in DbgMCU_APB1_Fz register (example: using the .dbgconf debug configuration file) will prevent TIM14 to run when the core is halted and avoid the issue reported.

jreineckearm, 2025-06-04:
Can confirm that setting freeze registers mitigates the impact of this HW defect for the STM32F7 series.
It could be an option to change dbgconf defaults to freeze peripherals. Could imagine users prefer to disable the freeze when needed rather than starting simple device usage with unstable debug.

KeilChris, 2025-06-05:
Can someone share the dbgconf here for testing purposes, please?

KeilChris, 2025-06-05:
Works as expected, thanks for this!

jreineckearm, 2025-07-14:
Closing issue, we still have #159 for documenting this/the workaround.

ReinhardKeil, 2025-07-15:
For a plausible user documentation this is required https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/issues/397