Open-CMSIS-Pack/vscode-cmsis-debugger #259: Enable Local Markdown Linting to Reduce CI Usage

soumeh01, 2025-05-19:
Currently, markdown linting runs only as part of the CI process. As a result, developers must wait for the CI to complete before addressing any markdown-related issues, which introduces delays and consumes unnecessary CI minutes.

### Considerations:

- Investigate the feasibility of running markdown linting locally.
- Provide developer instructions for local linting (e.g., via a script or documented CLI usage).
- Ensure local results are consistent with CI markdown linting rules.
- CI should still retain the markdown linting step as a backup check.

### Benefits:

- Faster feedback for developers.
- Reduced CI consumption and costs.
- Improved developer experience.

soumeh01, 2025-05-20:
The commands have been introduced and can be used locally to validate changes in the documents. Here are the instructions to be followed https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/blob/main/DEVELOPMENT.md#updating-documentation

closed 2025-05-20 by soumeh01

jreineckearm, 2025-05-20:
Thanks, @soumeh01 !