Open-CMSIS-Pack/vscode-cmsis-debugger #260: [Feature Request]: PERIPERAL View is named RUN AND DEBUG in secondary view

ReinhardKeil, 2025-05-19:
### Description

When the  PERIPERAL View is moved to the secondary bar, it is named RUN AND DEBUG.
WATCH View works as expected.

[image]

### Is this request a Security Requirement?

- [ ] Yes

### Priority

Medium

### Additional Notes (Optional)

CMSIS Debugger Version 0.20

jreineckearm, 2025-05-20:
@thorstendb-ARM , please investigate after the GDB adapter work if we can do something about it.

thorstendb-ARM, 2025-05-21:
Other comments:
- should all error messages be duplicated to popups? 
[image]
-

thorstendb-ARM, 2025-05-21:
Found that the standard behavior of draged-out views is, that they inherit the containr name (which is "Run and Debug" for the "Debug" Container.
Sometimes (seems to be very fragile) the view name gets added, e.g.: "Run and Debug:Peripherals".
Tested creating my own container named "Peripherals" with just one view "Peripherals" in it. Most time it comes up as "Peripherals", but when docked into "Run and Debug", it shows up as "Peripherals:Peripherals".
When relating it to the "Activity bar" (Sidebar where Run, CMSIS etc. live), without a surrounding container, it seems to just display as "Peripherals" and can be docked into other containers without getting a strange name. For this, an icon must be set which should appear in the Activity Bar but I didnÄt get this to be shown (just the Perpherals window).

jreineckearm, 2025-06-23:
Further investigation confirms a lot of the above.

Short version:
I'd suggest to create a view container in the "Panel" (bottom) area with the same title as the view as. Then register the view to the new container. You cannot create a view container with the right side area as default location.

Longer version:
Creating a new view container sounds like the right thing to do. The view container name (in our case "Run and Debug") shows if you move a view to an empty area. That's intended behavior of VS Code to show the logical grouping by container. Even if it creates "duplicates".
But you can only use the following as parents
* "Activity Bar": as done for "Run and Debug" and "CMSIS". It shows initially as icon on the left.
* "Panel": the bottom area, as done for "Serial Monitor". Initially hides a provided icon, but not sure if much better because even less vertical space for the peripherals tree.
* You cannot register a view panel is child of the "sidebar" on the right side. This seems to be "reserved" as a kind of user space.

In any case, you should contribute an icon for the view container. It shows if moved to the activity bar, but not into another view container. If moved into another view container, it shows the "container:view" label to mark it as a "guest".

ReinhardKeil, 2025-07-15:
Postponed as it needs an overall concept for the Live Views