Open-CMSIS-Pack/vscode-cmsis-debugger #263: [Feature Request]: Improve Robustness in case of user errors

ReinhardKeil, 2025-05-19:
### Description

pyOCD should be tested in typical failure scenarios.

For example when target hardware is not powered.  Currently it shows:

[image]

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

jreineckearm, 2025-05-20:
Main work expected in pyOCD. But may need some investigation on extension/VS Code side to show a more meaningful error pop-up. What you currently see is VS Code default behavior in case of a failing task process (shows its exit code).

ReinhardKeil, 2025-07-07:
see #370