Open-CMSIS-Pack/vscode-cmsis-debugger #267: [Feature Request]: Option to suppress pyOCD CoreSight discovery output

TeoMahnic, 2025-05-20:
### Description

When launching the `gdbserver` using pyOCD, the CoreSight discovery output is currently always printed to the console. This can result in excessive and unnecessary logging.

Several other subcommands such as `load` and `erase` already support a quiet or suppressed output mode. Introducing similar behavior for the `gdbserver` would enhance consistency and provide better control over output verbosity.

### Is this request a Security Requirement?

- [ ] Yes

### Priority

Low

TeoMahnic, 2025-07-28:
Related to https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/issues/397

jreineckearm, 2025-07-30:
@thorstendb-ARM , could you please help with validating the changes Teo made to offload @ReinhardKeil ?
Please use the pre-release VSIX here (N/A on marketplace): https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/releases/tag/v1.0.0-pre0

RobertRostohar, 2025-07-31:
Note that the following options need to be added the launch/task,json for pyocd gdb server:
```json
"--quiet",
"--log", "*.cbuild_run,*server=info"
```

This will be part of the templates in the upcoming CMSIS Solution extension.

thorstendb-ARM, 2025-07-31:
Tested: 
- with "--quiet" no pyOCD messages are shown anymore

RobertRostohar, 2025-07-31:
This is what we expect at this point.

There should be also a line quite early about the target, example:
`0002224 I Target device: STM32H743ZITx [cbuild_run]`