Open-CMSIS-Pack/vscode-cmsis-debugger #286: [Bug]: Better handling of pyOCD leftover processes

KeilChris, 2025-05-27:
### Description

When a debug session crashed unexpectedly or continues running without notice, a pyOCD process will continue running which makes consecutive debug sessions impossible.
For example, when I forget to add the GDB toolchain to my project, I get the error that "gdb is not running" and it seems that the whole session exits. But still, pyOCD is running in the background. When I start another session, I get strange errors that do not tell me the root cause of the problem (for example "Access denied(insufficient permissions)) or "cannot do this when in exec mode").

### Steps to Reproduce

1. Start a debug session without Arm GCC installed.
2. Wait for "gdb is not running"
3. Correct the `vcpkg-configuration.json` file and use Arm GCC.
4. Start a new debug session: "Access denied".

jreineckearm, 2025-05-27:
Expected resolution for #89 and https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter/issues/361 should mitigate effects in some scenarios already. Need to investigate what other measures to implement beyond that and where.

thorstendb-ARM, 2025-05-27:
Also having this issue.
I'd suggest to extend pyOCD with a watcher-function which looks if any gdb and/or target-debugger is connected, and exit if not.

jreineckearm, 2025-07-07:
PR opened here: https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter/pull/392 This does not yet address better error messages to keep the PR size reviewable.

jreineckearm, 2025-07-14:
Should have improved with CDT GDB Adapter 2.1.0. @KeilChris , please give it a try and close if better.