Open-CMSIS-Pack/vscode-cmsis-debugger #288: [Feature Request]: pyOCD list --cbuild-run <file> should provide information about debug connection

ReinhardKeil, 2025-05-27:
### Description

`pyOCD list` shows the available debug adapters
`pyOCD list --cbuild-run <file>` should show details about the connection to the target system (similar to the uVision Debug options dialog).

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

RobertRostohar, 2026-10-02:
@ReinhardKeil 

The proposal is to introduce a new `pyocd info` sub-command that provides the information about the connected target rather than using `pyocd list` which is meant to display available probes (and not actual target connections).

The `info` sub-command displays information about debug probe (including unique id - needs to be selected if there are multiple probes), debug port type and clock with CPU information of detected core(s).

The existing VS Code task `CMSIS TargetInfo` would be changed to execute the following command:
`pyocd info --probe <id> --cbuild-run <file>`

This is aligned with other CMSIS tasks that all operate on a target behind selected probe.

Example of an output for a single core:
```
Debug Probe: STM32 STLink [0669FF505352716587121328]
Debug Port: SWD [1MHz]
CPU core: Cortex-M33 r0p2, v8.0-M architecture
  Extensions: DSP, FPU, FPU_V5, MPU
  8 hardware breakpoints, 1 literal comparators
  4 hardware watchpoints
```

Example of an output for a multi core:
```
Debug Probe: STLINK-V3 [003D004D3137511133333639]
Debug Port: SWD [10MHz]
CPU core #0: Cortex-M7 r1p1, v7.0-M architecture
  Extensions: DSP, FPU, FPU_DP, FPU_V5, MPU
  8 hardware breakpoints, 1 literal comparators
  4 hardware watchpoints
CPU core #1: Cortex-M4 r0p1, v7.0-M architecture
  Extensions: DSP, FPU, FPU_V4, MPU
  6 hardware breakpoints, 2 literal comparators
  4 hardware watchpoints
```

When interested in more details the "-v" argument can be added. This will show additional information including CoreSight discovery.

Example:
```
0001771 I start-pname: CM4 [cbuild_run]
0001803 I Target device: STM32H747XIHx [cbuild_run]
0001878 I Target type is stm32h747xihx [board]
0002110 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0002639 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0002642 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0002645 I APB-AP#2 IDR = 0x54770002 (APB-AP var0 rev5) [discovery]
0002649 I AHB-AP#3 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
0002652 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=020:ST part=450) [rom_table]
0002653 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c7> [rom_table]
0002654 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c7) [rom_table]
0002655 I   [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0002656 I   [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0002657 I   [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=00e> [rom_table]
0002658 I   [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0002659 I [1]<e0041000:ETM M7 class=9 designer=43b:Arm part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0002661 I [2]<e0043000:CTI CS-400 class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0002662 I APB-AP#2 Class 0x1 ROM table #0 @ 0xe00e0000 (designer=020:ST part=450) [rom_table]
0002664 I [2]<e00e3000:SWO CS-400 class=9 designer=43b:Arm part=914 devtype=11 archid=0000 devid=ea0:0:0> [rom_table]
0002665 I [3]<e00e4000:Trace Funnel CS-400 class=9 designer=43b:Arm part=908 devtype=12 archid=0000 devid=32:0:0> [rom_table]
0002666 I [4]<e00e5000:TSGEN class=15 designer=43b:Arm part=101> [rom_table]
0002667 I [5]<e00f0000:ROM class=1 designer=020:ST part=001> [rom_table]
0002667 I   APB-AP#2 Class 0x1 ROM table #1 @ 0xe00f0000 (designer=020:ST part=001) [rom_table]
0002670 I   [0]<e00f1000:CTI CS-400 class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0002672 I   [2]<e00f3000:Trace Funnel CS-400 class=9 designer=43b:Arm part=908 devtype=12 archid=0000 devid=34:0:0> [rom_table]
0002674 I   [3]<e00f4000:ETF class=9 designer=43b:Arm part=961 devtype=32 archid=0000 devid=380:0:0> [rom_table]
0002675 I   [4]<e00f5000:TPIU CS-400 class=9 designer=43b:Arm part=912 devtype=11 archid=0000 devid=a0:0:0> [rom_table]
0002677 I AHB-AP#3 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=450) [rom_table]
0002679 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0002680 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0002680 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0002681 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0002683 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0002684 I [6]<e0043000:CTI CS-400 class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0002693 I CPU core #0: Cortex-M7 r1p1, v7.0-M architecture [cortex_m]
0002693 I   Extensions: [DSP, FPU, FPU_DP, FPU_V5, MPU] [cortex_m]
0002693 I   FPU present: FPv5-D16-M [cortex_m]
0002700 I CPU core #1: Cortex-M4 r0p1, v7.0-M architecture [cortex_m]
0002700 I   Extensions: [DSP, FPU, FPU_V4, MPU] [cortex_m]
0002700 I   FPU present: FPv4-SP-D16-M [cortex_m]
0002702 I core 0: Cortex-M7 r1p1, pname: CM7 [cbuild_run]
0002702 I core 1: Cortex-M4 r0p1, pname: CM4 [cbuild_run]
0002704 I 4 hardware watchpoints [dwt]
0002708 I 8 hardware breakpoints, 1 literal comparators [fpb]
0002723 I 4 hardware watchpoints [dwt]
0002727 I 6 hardware breakpoints, 2 literal comparators [fpb]
Debug Probe: STLINK-V3 [003D004D3137511133333639]
Debug Port: SWD [4MHz]
CPU core #0: Cortex-M7 r1p1, v7.0-M architecture
  Extensions: DSP, FPU, FPU_DP, FPU_V5, MPU
  8 hardware breakpoints, 1 literal comparators
  4 hardware watchpoints
CPU core #1: Cortex-M4 r0p1, v7.0-M architecture
  Extensions: DSP, FPU, FPU_V4, MPU
  6 hardware breakpoints, 2 literal comparators
  4 hardware watchpoints
```

RobertRostohar, 2026-10-08:
`pyocid info` sub-command has been added with PR https://github.com/pyocd/pyOCD/pull/2040

RobertRostohar, 2026-10-08:
https://github.com/Open-CMSIS-Pack/debug-adapter-registry/pull/125
Debug Adapter registry update to use the new `pyocd info` command

TeoMahnic, 2026-10-14:
Verified with pyOCD v0.45.2.dev71+gd6a9f5d
```
pyocd info --cbuild-run "out\Blinky+STM32H745I-DISCO.cbuild-run.yml"
Debug Probe: STLINK-V3 [002B00203137511539383538]
Debug Port: SWD [4MHz]
CPU core #0: Cortex-M7 r1p1, v7.0-M architecture
  Extensions: DSP, FPU, FPU_DP, FPU_V5, MPU
  8 hardware breakpoints, 0 literal comparators
  4 hardware watchpoints
CPU core #1: Cortex-M4 r0p1, v7.0-M architecture
  Extensions: DSP, FPU, FPU_V4, MPU
  6 hardware breakpoints, 2 literal comparators
  4 hardware watchpoints
```

closed 2026-10-14 by TeoMahnic