Open-CMSIS-Pack/vscode-cmsis-debugger #303: [Feature Request]: Rename "Custom Reset" to "Reset Target", keybinding

ReinhardKeil, 2025-06-05:
### Description

Suggest to rename "Custom Reset" to "Reset Target"

[image]

A potential default key binding could be Ctrl + Alt + F5

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

thorstendb-ARM, 2025-06-05:
Reason for naming:
The whole procedure (from launch.json through source code up to the button) has been named "custom reset", because the button just executes the instructions added to launch.json.
The feature is very generic and not bound to a "Reset Target" but can be anything a user can put there. So, imho we should not go for "Reset Target", as a user can do (reset) anything there.

thorstendb-ARM, 2025-06-10:
Changed as requested:
[image]

Did not add the key binding because on Mac CMD+OPT+F5 is used by MacOS for accessibillity configuration.

see PR https://github.com/eclipse-cdt-cloud/cdt-gdb-vscode/pull/160

jreineckearm, 2025-06-17:
Reopen, PR not merged yet and can't see key binding addressed.
Assigning to myself until completed.

jreineckearm, 2025-06-18:
Raised https://github.com/eclipse-cdt-cloud/cdt-gdb-vscode/issues/161 for tracking in the Elipse CDT.cloud repo.

jreineckearm, 2025-06-23:
Could go for something more complex like `Ctrl+Shift+Alt+R` which isn't used yet. Or a sequence like `Ctrl+K T` (the T without Ctrl), which is only used in the context of Notebooks. Which are probably less likely to conflict with a debug session.

But also happy to park and see how users get along without it.

jreineckearm, 2025-06-24:
Skipped keybinding part as agreed offline. Let's create a new issue if relevant again in future.
Comes with next CDT GDB Adapter release (2.0.6 and later).

jreineckearm, 2025-07-02:
Change now released in CDT GDB Adapter extension 2.0.6.