Open-CMSIS-Pack/vscode-cmsis-debugger #316: [Bug]: AC6 Stack Window fails

ReinhardKeil, 2025-06-11:
### Description

The optimization of LR register in AC6 are not considered in the GDB stack unwinder. 

In uVision Debugger, the root addresses of __Reset_Handler and threads are known. This stops the stack unwinder.

In GDB this implementation is likely impossible. More investigation is required.

ReinhardKeil, 2025-07-02:
Unfortunately the `backtrace limit` and other solutions shown above do not work (yet).

RobertRostohar, 2026-09-10:
After quite some investigation, we know a bit more about this.

When functions have the `noreturn` attribute, then Arm compiler 6 (and also armclang) uses/destroy the LR but does not emit the debug information `.cfi_undefined 14`. This seems to be an issue in the compiler. We are waiting for confirmation.

gdb unwinder has therefore many problems including not correctly unwinding stack or even worse in some circumstances (no optimization) where the stack unwinder blocks indefinitely or for a long time.

There is a compiler command line option `-funwind-tables` which seems to overcome this problem. The generated code will preserve the LR on stack (slightly higher stack usage) but stack unwinding then works correctly and reliable.

For the time being, it is recommended to add `-funwind-tables` command line option to the AC6/armclang when building for debugging.

ReinhardKeil, 2026-09-15:
@RobertRostohar could you add a link to the issue in the CLANG project to simplify tracking?