Open-CMSIS-Pack/vscode-cmsis-debugger #347: [Bug]: FreeRTOS kernel awareness in pyOCD does not work

ReinhardKeil, 2025-06-27:
### Description

This project [Blinky_HE3.zip](https://github.com/user-attachments/files/20940254/Blinky_HE3.zip) does not show the running threads using CMSIS-DAP.

[image]

jreineckearm, 2025-07-28:
Not realistic to solve for 1.0.0

DavidLesnjak, 2025-08-01:
This issue is complex, requiring concurrent analysis of the main and feature/freertos_10 branches and the FreeRTOS implementation, so it will take some time.

ReinhardKeil, 2026-10-21:
A public reproducer using CMSIS-FreeRTOS 11.3.0 on an Alif DevKit-E7 is available here:

https://github.com/ReinhardKeil/Alif-FreeRTOS

With CMSIS-DAP@pyOCD, the GDB-server output reports:

```
FreeRTOS: pxReadyTasksLists size is unexpected, maybe an unsupported version of FreeRTOS. Try using the --elf option.
```

This appears after the application has started and affects FreeRTOS kernel/thread awareness rather than application execution. The project uses the SCVD from `ARM::CMSIS-FreeRTOS@11.3.0` and provides a concrete current test case for this issue.

The project also reproduces the independent second-debug-start failure tracked in #999; power-cycling the board restores debugging.