Open-CMSIS-Pack/vscode-cmsis-debugger #349: [Bug]: Failed in GDB version check

syamknair, 2025-06-27:
### Description

When launching the debugger using vscode in a CMSIS toolbox on a WIndows 11 box,  I get the following error message:

[image]

I verified that the arm-none-eabi-gdb exists on my machine and it is available in the cmsis toolbox environment

[image]

I am using the GDB that came with GCC installation of the ARM tools environment manager:

Here is my launch configuration:

{
            "name": "CMSIS Debugger: pyOCD",
            "type": "gdbtarget",
            "request": "launch",
            "cwd": "${workspaceFolder}",
            "program": "${command:cmsis-csolution.getBinaryFile}",
            "gdb": "arm-none-eabi-gdb",
            "initCommands": [
                "load",
                "tbreak main"
            ],
            "target": {
                "server": "pyocd",
                "port": "3333"
            },
            "cmsis": {
                "cbuildRunFile": "${command:cmsis-csolution.getCbuildRunFile}"
            }
}

I am using a CMSIS-DAP debugger and am able to successfully program the device. Failure happens during a debug session.

### Steps to Reproduce

1. Create a CMSIS toolbox project for a known device
2. Create a launch configuration
3. Attach the target using a CMSIS-DAP debugger
4. Try to debug the code and gets the above error

tim-lip, 2025-07-01:
it's a v1.x device

tim-lip, 2025-07-01:
When I run a "load & run" I get:

```
Executing task: pyocd load --probe cmsisdap: --cbuild-run c:\Users\M76685\MPLABProjects\Keil\SAMC21-Keil\SAMC21-Keil+ATSAMC21J18A.cbuild-run.yml 

0000757 I Loading C:\Users\M76685\MPLABProjects\Keil\SAMC21-Keil\out\test-project\ATSAMC21J18A\Debug\test-project.axf [load_cmd]
[==================================================] 100%
0001107 I Erased 16384 bytes (1 sector), programmed 1024 bytes (1 page), skipped 0 bytes (0 pages) at 2.86 kB/s [loader]
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: pyocd gdbserver --probe cmsisdap: --connect attach --persist --reset-run --cbuild-run c:\Users\M76685\MPLABProjects\Keil\SAMC21-Keil\SAMC21-Keil+ATSAMC21J18A.cbuild-run.yml 

0000609 I Target type is atsamc21j18a [board]
0000662 I DP IDR = 0x0bc11477 (v1 MINDP rev0) [dap]
0000678 I AHB-AP#0 IDR = 0x04770031 (AHB-AP var3 rev0) [discovery]
0000678 I AHB-AP#0 Class 0x1 ROM table #0 @ 0x41003000 (designer=01f part=cd0) [rom_table]
0000678 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c0> [rom_table]
0000678 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c0) [rom_table]
0000678 I   [0]<e000e000:SCS v6-M class=14 designer=43b:Arm part=008> [rom_table]
0000678 I   [1]<e0001000:DWT v6-M class=14 designer=43b:Arm part=00a> [rom_table]
0000694 I   [2]<e0002000:BPU v6-M class=14 designer=43b:Arm part=00b> [rom_table]
0000696 I [1]<41008000:MTB M0+ class=9 designer=43b:Arm part=932 devtype=31 archid=0a31 devid=0:0:0> [rom_table]
0000699 I CPU core #0: Cortex-M0+ r0p1, v6.0-M architecture [cortex_m]
0000699 I   Extensions: [MPU] [cortex_m]
0000699 I Setting core #0 (Cortex-M0+) default reset sequence to ResetSystem [cbuild_run]
0000699 I 2 hardware watchpoints [dwt]
0000699 I 4 hardware breakpoints, 0 literal comparators [fpb]
0000827 I Semihost server started on port 4444 (core 0) [server]
0000861 I GDB server started on port 3333 (core 0) [gdbserver]
```

I am using SAMC21J18A xplained pro board here.

KeilChris, 2025-07-01:
That looks correct! So you "only" have problems when entering debug?

ReinhardKeil, 2025-07-01:
I have tested with such a board on my computer. Everything worked smoothly.  I have used the Arm Environment Manager to install GCC.  How did you install GCC/GDB.  Could you send us your PATH variable setting?

ReinhardKeil, 2025-07-01:
@syamknair what target are you using? Is your CMSIS-DAP adapter running firmware version 1 (HID) or version 2 (WinUSB) communication?

tim-lip, 2025-07-01:
No problem, thank you for the support.

No I removed the other toolchain 
```
> where.exe arm-none-eabi-gdb
c:\Users\Username\.vcpkg\artifacts\2139c4c6\compilers.arm.arm.none.eabi.gcc\14.2.1\bin\arm-none-eabi-gdb.exe
```
and the error still persists.

My launch config:
```
{
            "name": "CMSIS_DAP@pyOCD (launch)",
            "type": "gdbtarget",
            "request": "launch",
            "cwd": "${workspaceFolder}/",
            "program": "out/test-project/ATSAMC21J18A/Debug/test-project.axf",
            "gdb": "arm-none-eabi-gdb",
            "preLaunchTask": "CMSIS Load",
            "initCommands": [
                "monitor reset halt",
                "tbreak main"
            ],
            "customResetCommands": [
                "monitor reset halt"
            ],
            "target": {
                "server": "pyocd",
                "serverParameters": [
                    "--probe",
                    "cmsisdap:",
                    "--connect",
                    "attach"
                ],
                "port": "3333"
            },
            "cmsis": {
                "cbuildRunFile": "${command:cmsis-csolution.getCbuildRunFile}",
                "updateConfiguration": "auto"
            }
        }
```

Let me know if I can provide any additional information.

syamknair, 2025-07-01:
I am using an atmel debugger as well

[image]

The error is when the cmsis toolbox extension  is launching gdb. 

I was able to launch pyocd with this debugger and gdb on two different command line instances and then step through the code.

jreineckearm, 2025-07-02:
@tim-lip , @syamknair , in order to find out what's going wrong I have replicated the call to get the GDB version in a test version of the CMSIS Debugger extension.

[image]

The output shows in the output windows:

[image]

Would you mind downloading and installing a dev drop from [here](https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/actions/runs/16019211067), and execute the two commands from your workspace that is failing to launch the debugger.
(Note that the dev drop may disappear after a couple of days due to retention time for those build artifacts. Please ping me if that happens)

In a first instance, I'd like to confirm that the problem also shows with this (the command without `via Shell`). Secondly, I'd like to see if it behaves differently if requesting the command to go through a shell (the command with `via Shell`). Please call the commands via the command palette (CTRL+SHIFT + P).

To install the above linked dev drop, download and unzip the [file](https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/actions/runs/16019211067/artifacts/3447160300) for Windows. The install it as explained [here](https://code.visualstudio.com/docs/configure/extensions/extension-marketplace#_install-from-a-vsix).

Thanks for your help with this.

syamknair, 2025-07-02:
@jreineckearm 

Here is the output I see:

[image]

jreineckearm, 2025-07-02:
@syamknair , thanks for trying this and confirming that this problem isn't only showing in the CDT GDB adapter! This gave me more confidence in where to continue investigating. Much appreciated!

I still cannot fully exclude security related issues. But I agree it's less likely when you don't see this with other frameworks.

I've done more investigation in the meanwhile. And there is a chance that the way the child process is called might cause problems. I've replaced it with another approach in the previously shared reproducer. A new test version can be found [here](https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/actions/runs/16030028888) .

Could you please try again? Same commands, just slightly different output (also showing the length of captured stdout/stderr output).

syamknair, 2025-07-02:
@jreineckearm , here you go:

[image]

jreineckearm, 2025-07-02:
Thanks for the swift response, @syamknair ! Again much appreciated, even if not with the hoped for result. :-(
We'll continue with the investigation.

jreineckearm, 2025-07-03:
@syamknair , would you mind running another test to help us diagnose the root cause?
I've updated the test extension. It now has the following commands

[image]

The output for the `Spawn` versions generate now more details about the created child process. Could you please run
`Get GDB Version (TEST) - Spawn` and `Get GDB Version via Shell (TEST) - Spawn`?

New test version can be found [here](https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/actions/runs/16047669796).

One other thing worth a try is to replace the launch configuration setting `"gdb": "arm-none-eabi-gdb"` with the absolute path to the executable that works from the terminal (`"gdb": "<your_executable_path>"`). And try again to connect.

Thanks a lot again for your help.

syamknair, 2025-07-03:
Here are the execfile output

[image]

jreineckearm, 2025-07-03:
Thanks again, @syamknair ! So, it's not necessarily a problem with the method of calling it. But the executable itself behaves differently than for example `cbuild` or `where`. We'll dig into the differences....

ReinhardKeil, 2025-07-03:
@syamknair thanks for all your help.  We still don't understand `arm-none-eabi-gdb` version check fails.

`arm-none-eabi-gdb` uses some environment variables that may impact the operation.  Are any of these variables set on your computer?

Variable | Description
-- | --
HOME | Used to locate ~/.gdbinit.
GDBINIT | Overrides the default initialization file.
PYTHONPATH | Used if your GDB build supports Python scripting.
PATH | Used to locate helper tools like arm-none-eabi-objdump, arm-none-eabi-nm, etc.
TERM, LINES, COLUMNS | Control terminal display behavior.
LANG, LC_ALL, etc. | Control localization.
GDBHISTFILE, GDBHISTSIZE | For command history handling.

syamknair, 2025-07-03:
@ReinhardKeil  @jreineckearm 

-   I verified that those environment variables are not set in my machine.
-  I ran the same steps on my colleague's workstation with the same results
-  I ran vscode as administrator, setup the extensions again and ran it and had the same results

I am also leaning towards the theory that some corporate security software is blocking the launch of arm-none-eabi-gdb from launching correctly from vscode extension. I will also check with our corp IT security team as well.

syamknair, 2025-07-04:
Hi @ReinhardKeil , @jreineckearm 
Checked internally within our organization and looks like the security software that we run on our company issued laptop do have issues when using node's process APIs to launch an executable. 
We were able to bypass the restrictions by making the tool aware of the executable. 
I am not sure If I am allowed to share more details about the fix or tools in question in a public forum, but we will track this internally and let other internal users know of the workarounds.