Open-CMSIS-Pack/vscode-cmsis-debugger #382: [Bug]: CMSIS-DAP v1.x using HID fails on macOS

KeilChris, 2025-07-10:
### Description

When I use a CMSIS-DAP v1.x compliant debug probe (which uses USB HID), it fails with:
```txt
 *  Executing task: pyocd load --probe cmsisdap: --cbuild-run /Users/chrsei01/03_work/02_Projects/Alif/AppKit/Blinky_HE/Blinky+AppKit-E7.cbuild-run.yml 

0010773 C Error: already open [__main__]
```
This is a timeout due to a non-working USB connection.

### Steps to Reproduce

1. Have the tools installed on a Mac.
2. Create a project.
3. Try to debug using a CMSIS-DAP v1.x probe, for example ULINKplus with v1.3.

### Expected Behavior

Debugging should show no errors and should work out of the box.

### Additional Information

Checking other Arm debug tools, it seems that there is a general problem with the libusb library under macOS. Others have overcome this by using HIDAPI (https://github.com/signal11/hidapi). Maybe this is the solution for this issue as well.

DavidLesnjak, 2025-07-25:
Resolved with https://github.com/pyocd/pyOCD/pull/1792

KeilChris, 2025-07-25:
Tested with a separate build. Works as expected now.

closed 2025-07-25 by KeilChris