Open-CMSIS-Pack/vscode-cmsis-debugger #419: [Feature Request]: EventRecorder

flair0626, 2025-07-23:
### Description

How to use the EventRecorder printf function of the Arm Keil Studio Pack (MDK v6) in VS Code？

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

jkrech, 2025-07-28:
At this moment there is no graphical output for the Event Recorder in the VSCode CMSIS Debugger extension.
As a work around the EventRecorder Component Variant named `Semihosting` can be selected (CMSIS-View:Event Recorder&Semihosting) needs to be selected and the application build for Semihosting. The application must be executed with the debugger attached. In the project directory a file named EventRecorder.log is created.
Run the utility `eventlist` to process the eventlog file and display results in the terminal.
See: https://github.com/ARM-software/CMSIS-View/tree/main/tools/eventlist#readme
`eventlist` is distributed as part of the Arm distribution of the CMSIS-Toolbox via the Arm Environment Manager extension, vcpkg or downloaded from here:
latest: https://artifacts.tools.arm.com/cmsis-toolbox/2.10.0

KeilChris, 2025-07-29:
I have created https://github.com/ARM-software/CMSIS-View/pull/145 to explain how to use semihosting in Keil Studio to gather Event Recroder data.