Open-CMSIS-Pack/vscode-cmsis-debugger #443: [Feature Request]: Show CPU execution time and provide stopwatch features

jreineckearm, 2025-08-04:
### Description

Depending on CPU/system implementation you can have HW resources to measure the CPUs execution time. One example is the v7-M/v8-M DWT CYCCNT register.

Such cycle counter information should in first step be collected and displayed.
And in a second step we should build "stopwatch" features around this. For example to measure the execution time between two breakpoints.

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

ReinhardKeil, 2025-08-11:
A potential user interface could be:

Add to Status bar

[image]

CPU is a placeholder for the pname in multi-processor configurations.  Single core systems would not show CPU. 

pname changes with the core selection in the debug toolbar

When the time cannot be calculated, the output is in states, i.e. `187129312 states`
On click to the status bar symbol open the command window with

jreineckearm, 2025-08-20:
https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter/issues/425 no longer urgent to resolve this. Extracting stop location now from stacktrace information polled on stopped thread/target.