Open-CMSIS-Pack/vscode-cmsis-debugger #473: [Feature Request]: stop when packs are missing

ReinhardKeil, 2025-08-20:
### Description

When a `cbuild-run.yml` file refers packs and a command (i.e. load) is executed that requires algorithms of a missing packs:
 - write the missing packs to a file `packs.cmd`.  The format is one pack per line (identical with the --missing option of csolution).
 - output error message as shown below

```txt
E packs with programming algorithm missing. Install with `cpackget add -f packs.cmd` 
```

See also https://open-cmsis-pack.github.io/cmsis-toolbox/build-tools/#install-missing-packs

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

TeoMahnic, 2025-08-27:
Addressed in https://github.com/pyocd/pyOCD/pull/1803

jreineckearm, 2025-09-10:
Nice, works as expected.
Deliberate choice to write `packs.txt` instead of `packs.cmd`.
Copy & paste the command from the error message works seamlessly.

closed 2025-09-10 by jreineckearm