Open-CMSIS-Pack/vscode-cmsis-debugger #487: [Feature Request]: Question: How to handle breakpoints to common code in multi-processor systems

ReinhardKeil, 2025-09-03:
### Description

A multi-processor system may implement the same code on different cores in a multi-core system.

How can I as a user select the core to set a breakpoint in such a module to just one processor.

Example is the MCX multi-core example `vioSetSignal` function.

[image]

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

jreineckearm, 2025-09-03:
Related to #329

jreineckearm, 2025-10-03:
Breakpoints in VS Code are considered global and to apply to all sessions. Any change outside of VS Code to manage additional session ("debug connection") info for breakpoints will become messy and create a disconnect from what's shown in the GUI. Hence needs extensions in VS Code.

Unscheduled feature request for VS Code since May 2023: https://github.com/microsoft/vscode/issues/181794