Open-CMSIS-Pack/vscode-cmsis-debugger #538: [Feature Request]: pyOCD CI support similar to FVP

ReinhardKeil, 2025-09-29:
### Description

pyOCD should support a seamless CI integration for regression testing on hardware.

# Features:
 
- installation only requires download algorithms (these are typically provided in packs).
  - missing packs are tolerated as long as the algorithms are there.  
- it should be sufficient to provide the content of the `out` directory 
   - missing .cmsis directory with *.dbgconf is tolerated
- similar behavior as FVP with options to:
   - redirect serial output to console
   - use EOT (0x04) to terminate run
   - specify an execution limit in seconds
   - specify an debug adapter ID

Provided that tools and flash algorithm is installed on a test computer, the CI workflow for RUN could have only 2 steps:

```yml
    - name: Download Build Artifact
      uses: dawidd6/action-download-artifact@v2
      with:
        workflow: Build_T2G_Release.yaml
        name: T2G-Kit@Release
        path: ./artifact/out

    - name: Flash-Download-Test for T2G-Kit@Release
      working-directory: ./artifact/out
      run: | 
         python3 -m pyocd load --probe cmsisdap: --cbuild-run T2G+T2G-Kit.cbuild-run.yml
```

Currently output (based on an example) is:
```
Run python3 -m pyocd load --probe cmsisdap: --cbuild-run T2G+T2G-Kit.cbuild-run.yml
0000503 W Missing required CMSIS packs. Install with 'cpackget add -f packs.txt' [cbuild_run]
0000943 I Loading /home/devadmin/runner-safety-ifx-t2g/_work/Safety-Example-Infineon-T2G/Safety-Example-Infineon-T2G/artifact/out/Boot/T2G-Kit/Release/Boot.hex [load_cmd]
[---|---|---|---|---|---|---|---|---|----]
[========================================]
0001341 I Erased 32768 bytes (1 sector), programmed 13824 bytes (27 pages), skipped 0 bytes (0 pages) at 35.09 kB/s [loader]
0001342 I Loading /home/devadmin/runner-safety-ifx-t2g/_work/Safety-Example-Infineon-T2G/Safety-Example-Infineon-T2G/artifact/out/TrafficLight/T2G-Kit/Release/TrafficLight.hex [load_cmd]
[---|---|---|---|---|---|---|---|---|----]
[========================================]
0004532 I Erased 163840 bytes (5 sectors), programmed 136704 bytes (267 pages), skipped 0 bytes (0 pages) at 44.29 kB/s [loader]
0004533 I Loading /home/devadmin/runner-safety-ifx-t2g/_work/Safety-Example-Infineon-T2G/Safety-Example-Infineon-T2G/artifact/out/Blinky/T2G-Kit/Release/Blinky.hex [load_cmd]
[---|---|---|---|---|---|---|---|---|----]
[========================================]
0004926 I Erased 32768 bytes (1 sector), programmed 14848 bytes (29 pages), skipped 0 bytes (0 pages) at 37.93 kB/s [loader]
```

Tasks:
  - identify how to close the gaps and provide a specification for implementation

### Is this request a Security Requirement?

- [ ] Yes

### Priority

High

jreineckearm, 2025-10-21:
To address soon after next release.

ReinhardKeil, 2025-11-03:
Proposal See #602